RoboJackets/urc-software #24: Implement Live Camera Feed

a-stickan, 2022-05-11:
## Description

Write nodes implementing a live camera-feed allowing for precise remote operation of the rover.

Because of the scale of this project and it being completely different to anything different we've done in IGVC, I'd recommend tackling this as a team and writing design documentation along the way. You may need to collaborate with urc-firmware and electrical to get this done.

## Requirements

 - [Camera is capable of the tasks outlined here](https://urc.marssociety.org/home/requirements-guidelines)
 - Design decisions are properly documented

pr05, 2022-07-09:
[image] [ci_pipeline](https://trello.com/c/ctbXuz5i/1-cipipeline)

a-stickan, 2022-07-17:
Finished with #44

closed 2022-07-17 by a-stickan