RoboJackets/urc-software #75: Automated build script

fourpenny, 2022-09-01:
## Description
Bash/zsh script to automate the [Ubuntu/Linux install instructions](https://github.com/RoboJackets/urc-software/blob/master/documents/installation/ubuntu_installation.md)

## Requirements
- .sh script

fourpenny, 2022-09-03:
@a-stickan 
Update - I've mostly finished the install script at this point, just need to figure out how to get added to the RoboNav repos so I can make a PR.

Something I actually would like your opinion on: I figured out the source of some of the issues we've been seeing on the command line where colcon-build errors out: these packages are not added to rosdep yet which is why people currently have to add them manually. 

As a temporary fix, I could either add these as a manual step in the install script (just loop through a list of the packages and install them using apt) or modify the rosdep rules file in the users'  /etc/ros/rosdep/sources.list.d sources file to point to a fork of the repo (either hosted by the team or a member) like is suggested in the documentation for contributing to rosdep: http://docs.ros.org/en/independent/api/rosdep/html/contributing_rules.html

Any thoughts?

a-stickan, 2022-09-03:
Personally I feel that the rosdep solution you proposed is a bit of a workaround. We should look for a solution that makes building simpler, not one that will require making forks and editing the rosdep. I think the Nanopb errors can be solved with some clever CMake and package logic, but I'll keep the rosdep idea in mind if that doesn't work out.

For non-Nanopb packages though, the standard rosdep commands should work. No idea why that would be dropping errors.

fourpenny, 2022-09-04:
@a-stickan 
Here's the rosdep file that I'm talking about:
https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml

The problems I mentioned are basically just missing installing some of the system/apt packages when using rosdep because they aren't in the rosdep base.yaml file (which lists equivalent package sources for supported OS's)- these were the errors I got when originally following the setup instructions that I was able to fix by installing the packages manually at the command line. I think I will try to fix these by figuring out which ones are not included in rosdep right now and have bash download/install them in the install script I wrote.

I think these are different from the nanopb build errors that others were getting (I haven't had those errors yet so I'm not sure what they actually look like). I'll check out your PR from earlier to see if your fixed worked!

a-stickan, 2023-03-09:
Stale issue, closing

closed 2023-03-09 by a-stickan