RoboJackets/urc-software #99: test item

anshgandhi4, 2023-03-09:


rhkoulen, 2023-04-02:
This is a good source for elevation data:
https://gis.utah.gov/data/elevation-and-terrain/2018-lidar-southern-utah/
It has 0.25 sqm. pixel resolution (each pixel corresponds to a 0.5x0.5m square). Each data point has vertical accuracy ±0.1m.
To download the data:
- go to the interactive map from that page
- draw the region you want to fetch
- select the 0.5 meter DEM (DSM if available)
- download
It will be zipped in GeoTiff format. This is a rastered image file format. You can technically extract data directly, but there is a more practical way to retrieve the raw data:
- Download and open QGIS.
- Open the .tif in a new project.
- Raster>Conversion>Translate
- Under the field "Converted," specify the file you want to export to. The program will change the conversion based on the file extension you specify. gdal (the package built-in for conversion) is very finnicky, so it may or may not allow you to make a csv. If .csv is unavailable, .xyz is also very similar. Instead of comma separated, the x, y, and z values are space separated.

I'm not sure what the x and y values are yet. They're definitely in meters, which is probably sufficient for generating the costmap, but we want to have some sense of x-y to coordinates, so we can initialize the pose of the rover.

rhkoulen, 2023-04-09:
We already have the set of elevation points (see Issue #99) and NumPy has a vectorized framework to convert discretized elevation data to slope data.
```
granularity = 0.5 # our current best data has point granularity of 0.5 meters
elevation = np.array() # parse the .xyz elevation data into a NumPy array
x_grad, y_grad = np.gradient(elevation)
x_grad, y_grad = x_grad / granularity, y_grad / granularity
slope = np.sqrt(np.power(x_grad, 2) + np.power(y_grad, 2))
# Square root is not necessary.
# We may even want to transform these values during testing depending on the behavior of the robot.
# Just make sure any transformations are monotone.
```
Because it's vectorized, this _will_ be faster than any kernelization I would write. We can just use a ROS Python node to parse this data and create the static layer of the 2D costmap. I'm pretty sure we can just directly parse this data into the static layer. But how would that parsing look like? How is the static layer used? How can we use the 2D costmap to make a planned path? If we ever use vision, how do we import that dynamic obstacle layer into this costmap?

rhkoulen, 2023-04-09:
costmap_2d subscribes to a topic called "map" which takes an OccupancyGrid message.
At 3.1.2 [here](http://library.isr.ist.utl.pt/docs/roswiki/costmap_2d.html), this allows us to initialize the costmap from a user-generated static map. The formatting for the OccupancyGrid message is a little bit unclear. I don't understand the packages that we're using, but different ros documentations say that the data in the data array in OccupancyGrid are in different formats.

- uint8 [0-255] representing the cost of a cell
- uint8 [0-100] representing the percent likelihood that a given cell is an obstacle
- int8 (-1, 0, 1) -1: unknown, 0: not obstacle, 1: obstacle
- float [0-1] representing the likelihood that a given cell is an obstacle

I don't know how nav messages work, but this seems like a bit of an issue to me.

closed 2024-05-18 by davidcalderon03