RoboJackets/urc-software #211: Train YOLOv8 model to detect mallets and water bottles

yambati03, 2024-08-14:
One of the tasks in the autonomous missions dictate that we need to be able to identify mallets and water bottles in the environment. We want to fine-tune a YOLOv8 model to do this task. This includes curating a dataset from manual capture or open-source sets and writing infrastructure to train and run inference using the model.

yambati03, 2024-08-14:
@abhiramg2021 please add the links to the online datasets that you found here. Thanks!

abhiramg2021, 2024-08-14:
> @abhiramg2021 please add the links to the online datasets that you found here. Thanks!

Here are the three datasets I found on roboflow.

1.  https://universe.roboflow.com/mallet-and-water-bottle/mallet-and-water-bottle/model/5?webcam=true
2. https://universe.roboflow.com/wvu-urc-2024/wvu-urc-object-detection-data
3. https://universe.roboflow.com/michigan-mars-rover/mallet-bottle-detector/model/3

closed 2025-04-17 by mrinalTheCoder