RoboJackets/urc-software #252: Overhaul Simulation

mrinalTheCoder, 2025-02-27:
## Requirements

 - Switch to Gazebo Fortress from Gazebo Classic
 - Ensure ability to set spherical coordinates in Gazebo
 - Large terrain file
 - Functioning sensors in rover model
 - Rover URDF
 - Ability to run _all_ autonomy missions in sim

yambati03, 2025-03-02:
Also
* Be able to place ARUCO tags at arbitrary GPS coordinates
* Find appropriate models for the orange mallet and water bottle (see https://urc.marssociety.org/home/requirements-guidelines)
* For LiDAR, try using a GPU LiDAR plugin. CPU-only LiDAR plugins seem to be too slow, but experiment with both.

mrinalTheCoder, 2025-03-07:
Migration guide for switching ROS2 packages from Gazebo classic (Gazebo 11) to gazebo fortress: https://gazebosim.org/docs/latest/migrating_gazebo_classic_ros2_packages/

closed 2025-08-27 by mrinalTheCoder